radiasoft/rslaser_old #1: point to a Jupyter notebook that reads/writes SRW wavefront files

bruhwiler, 2020-06-29:
A link to another repo would be fine.
Alternatively, you can upload it to rslaser/examples/wavefronts/

bruhwiler, 2020-06-29:
Mod to initial comment:
The notebooks should go into
> rslaser/examples/notebooks

bnash, 2020-06-29:
For a notebook in an existing repository, here is a good example that creates a Gaussian wavefront in SRW. We use the low level python interface which allows more access to the data structures.
https://github.com/radiasoft/fastFELO/blob/master/wigner_function/Optics/FELObeamline4Xtal1to1_wignerFunction_gaussianBenchmark.ipynb

bnash, 2020-06-29:
I added an example notebook (and python file with a few functions):
https://github.com/radiasoft/rslaser/blob/master/rslaser/examples/notebooks/SRWwavefrontExample.ipynb

I saw there is another [examples folder](https://github.com/radiasoft/rslaser/tree/master/examples)
Should we put the example notebooks in the same directory?

bruhwiler, 2020-06-30:
I made a mistake and put my notebook in the wrong place.
The directory should be   rslaser/examples/notebooks/
Mine is in rslaser/rslaser/examples/...   which is wrong.

robnagler, 2020-06-30:
You can `git mv`

closed 2021-10-19 by bruhwiler