radiasoft/rslaser_old #6: create a demo notebook to illustrate solving the 3D diffusion equation in a cylinder

dtabell, 2020-11-30:
+ [ ] 3D diffusion equation with full BCs
+ [ ] 3D graphics
+ [ ] check scaling scaling
+ [ ] have simulation function accept sim parameters
+ [ ] supply notebook documentation
+ [ ] add cross-section plots
   - [ ] 1D
   - [ ] 2D

bruhwiler, 2022-03-14:
@dtabell -- can this be closed?

bruhwiler, 2022-04-30:
@dtabell -- please provide a link to the notebook and update the status of this issue.

bruhwiler, 2022-09-13:
Lowering priority. @dtabell -- please post a link to the notebook in question.

dtabell, 2023-01-03:
Multiple notebooks illustrate this. The latest, which includes the the nonlinear heat equation appears in
https://github.com/radiasoft/rslaser/blob/master/examples/thermal/NLThermalSaturation.ipynb

closed 2023-01-03 by dtabell