radiasoft/rslaser_old #27: errors in xt04_heat_TiAl2O3_3D.ipynb notebook

bruhwiler, 2021-10-29:
Execution of the first cell fails:

[image]

Seems like an incompatibility with the updated environment.

bruhwiler, 2021-10-29:
Maybe FEniCS needs to be reinstalled?

bruhwiler, 2021-10-29:
@robnagler  --  Dan is requesting that you have a look.

bruhwiler, 2022-04-30:
It no longer fails in the 1st cell, but now if fails in the 13th cell:

[image]

@dtabell please get this notebook working again.

Also -- do we need all of the notebooks in the xt04_heat_* series?  Please delete any that are not instructive.

dtabell, 2022-05-03:
The errors have been fixed.
Addressing a last few items before closing this issue.

bruhwiler, 2022-09-13:
@dtabell -- can this issue be closed?

bruhwiler, 2023-01-03:
Resolved.

closed 2023-01-03 by bruhwiler