radiasoft/rslaser_old #33: Rebuild FEniCS on Jupyter

dtabell, 2021-11-23:
Since restarting my Jupyter server, `from FEniCS import *` issues a raft of warnings.
(See below for a small sample of those.) I see that there have been several updates,
to JupyterLab and probably other parts.

```
/home/vagrant/.pyenv/versions/py3/lib/python3.7/site-packages/dolfin/jit/jit.py:121: RuntimeWarning: mpi4py.MPI.Status size changed, may indicate binary incompatibility. Expected 40 from C header, got 48 from PyObject
  def compile_class(cpp_data, mpi_comm=MPI.comm_world):
/home/vagrant/.pyenv/versions/py3/lib/python3.7/site-packages/dolfin/jit/jit.py:121: RuntimeWarning: mpi4py.MPI.Datatype size changed, may indicate binary incompatibility. Expected 24 from C header, got 32 from PyObject
  def compile_class(cpp_data, mpi_comm=MPI.comm_world):
/home/vagrant/.pyenv/versions/py3/lib/python3.7/site-packages/dolfin/jit/jit.py:121: RuntimeWarning: mpi4py.MPI.Request size changed, may indicate binary incompatibility. Expected 32 from C header, got 40 from PyObject
  def compile_class(cpp_data, mpi_comm=MPI.comm_world):
```

robnagler, 2021-11-23:
My guess is dolfin/fenics needs to be recompiled, because something changed in mpi/hdf5/numpy. You can also ignore the warnings for now. 

I'll rebuild fenics and get it on alpha today and report back here so you can test.

robnagler, 2021-11-23:
This is on alpha.sirepo.com/jupyter

nickgoldring, 2021-12-15:
Another thing to consider is that the meshing code we are currently using with FEniCS called mshr is deprecated and it is now recommended to use [gmsh](https://gmsh.info/) according to the developers:

[https://fenicsproject.discourse.group/t/mesh-density-parameter-in-generate-mesh-function/7301](https://fenicsproject.discourse.group/t/mesh-density-parameter-in-generate-mesh-function/7301)

dtabell, 2022-01-11:
@robnagler  The version you built on alpha works. Thank you!

As @nickgoldring mentions, the FEniCS community now deprecates `mshr` and recommends, `gmsh`.
Can you have a look at upgrading FEniCS to use Gmsh?

robnagler, 2022-01-11:
Please create a new issue here: https://github.com/radiasoft/download/issues/new

closed 2022-01-11 by robnagler

dtabell, 2022-01-11:
Done.