radiasoft/rslaser_old #35: parse laser profile images and smooth the 2D histogram

bruhwiler, 2021-12-17:
Please write Python code to parse two *.bmp files found here, https://github.com/radiasoft/rslaser/tree/master/rslaser/package_data/

The data should be smoothed to remove all small-scale artifacts.  Executable code should live here, https://github.com/radiasoft/rslaser/tree/master/examples

Any code encapsulated into libraries should live here, https://github.com/radiasoft/rslaser/tree/master/rslaser/utils

Time spent should be charged to 'silas' in GHG/clockwise.

kevin-bruhwiler, 2022-01-20:
I've pushed a couple of scripts into the examples/smoothing directory.

bruhwiler, 2022-01-29:
I renamed the scripts and modified the README

I added some notebooks using the smoothers and supergaussian fits.  I added a couple of notebooks that do explicitly Gaussian fits, so these can be compared to the supergaussian results.

bruhwiler, 2022-01-29:
@kevin-bruhwiler -- please add code that parses the following file, so the images can be plotted with correct spatial dimensions on the horizontal and vertical axes, instead of pixel number

https://github.com/radiasoft/rslaser/blob/master/rslaser/package_data/camera_sensor_info.txt

closed 2022-03-14 by bruhwiler