radiasoft/rslaser_old #38: Class CrystalSlice is defined twice

bnash, 2022-02-19:
CrystalSlice is defined [here](https://github.com/radiasoft/rslaser/blob/master/rslaser/elements/crystalSlice.py) and [here](https://github.com/radiasoft/rslaser/blob/master/rslaser/rscrystal/crystal.py).

I can see an argument for both `elements` and `rscrystal`. Maybe the tools for updating the crystal, such as the updates after slices, implementing Franz-Nodvik, etc should go in rscrystal, whereas the final propagation algorithm should go in `elements`?

gurhar1133, 2022-03-03:
Paul and I were thinking keeping all the crystal slice stuff in one place is better.

bruhwiler, 2022-03-14:
When this issue is closed, the elements/ directory should be deleted from the repository.

bruhwiler, 2022-03-14:
The relationship between Crystal() and CrystalSlice() is unclear.
Please create a separate issue for this and assign to you, me and Boaz.
Add the constructor item 4 from above to the new issue.

commit a02e5ef7 (2022-03-14): Fix #38: resolve drift class and nb fix
commit 5fd684bf (2022-03-15): Fix #38: propagate on element base
commit 7c113787 (2022-03-15): Merge pull request #57 from radiasoft/38-resolve-duplicate-defs

Fix #38 resolve duplicate defs
commit a1259adf (2022-03-15): Fix #38: moved CrystalSlice to elements with Element base
closed 2022-03-15 by bruhwiler