radiasoft/rslaser_old #39: initial tests for LaserPulse class

bruhwiler, 2022-03-01:
See example code here,
https://github.com/radiasoft/rslaser/blob/master/examples/notebooks/diagnostics/TransverseEnvelope.ipynb

* instantiate LaserPulse instance with zero chirp (i.e. d_lambda = 0), central wavelength lambda0=1e-6 meters and with 3 slices;  query the wavelength (i.e. _lambda0) associated with each slice; it should be the same as for the pulse as a whole
* do the same with d_lambda = lambda0/10; in this case, each slice should have a different wavelength, but probably won't because of a bug

gurhar1133, 2022-03-02:
After a couple small changes to the LaserPulse class, the first test passes. However, the second test fails. Is each slice's lambda0 supposed to be incremented or decremented by d_lambda during instantiation?

bruhwiler, 2022-03-02:
The code snippet above will result in each LaserPulseSlice instance having a photon energy that is inconsistent with the specified wavelength, that's getting passed in via the **_k argument.

I think all default values should be removed from the LaserPulseSlice constructor and the **_k argument should not be passed through from the LaserPulse() constructor.

This will result in the need for more syntax regarding management of arguments.

bruhwiler, 2022-03-02:
Perhaps a couple of tests for LaserPulseSlice() should be implemented before resolving the present issue.

Also, the PKDict object could be required to take photon energy (aka phE) rather than wavelength (aka lambda0).
Conversion back and forth between phE and lambda0 should be a shared function.

bruhwiler, 2022-03-02:
Given our agreement to use phE as the fundamental quantity and lambda as the derived quantity (which should not be added to the PKDict parameters), the d_lambda variable should also be abandoned and users should be required to specify _chirp.

commit d6d59fa2 (2022-03-02): Fix #39: wip
commit 357dcf74 (2022-03-02): Fix #39: test wip and small pulse refact
commit da46a704 (2022-03-02): Fix #39: WIP
commit 60816910 (2022-03-02): Fix #39: skeleton for validators and corresponding tests
closed 2022-03-03 by bruhwiler