radiasoft/rslaser_old #50: Set default values on LaserPulse

gurhar1133, 2022-03-07:
Rather than having constant declared separately with default parameter values, it would be better to have __Defaults set on the LaserPulse class itself. Then \_\_init\_\_ would fill in missing parameters using __DEFAULTS

Tests should be modified to throw exceptions when unknown parameters are fed to the \_\_init\_\_ 

LaserCavity may need these modifications as well

bruhwiler, 2022-03-07:
A related issue is that the default value for phE is not reasonable:
_PHE_DEFAULT = const.h * const.c / 1e-6

This assumes a 1 micron wavelength and converts it to energy, but the resulting units are in Joules.
Our units convention for photon energy is eV. Hence the correct convention would be:
_PHE_DEFAULT = const.h * const.c / const.e / 1e-6

Rather than using this equation, let's just specify the default as follows:
_PHE_DEFAULT = 1.55

bruhwiler, 2022-03-08:
Another somewhat related issue:
The default parameter
       sampFact=5,
in _LASER_PULSE_SLICE_DEFAULTS is creating unnecessarily large wavefront objects.

Please change it to
       sampFact=1,

commit 9b009262 (2022-03-07): Fix #50: defaults set on class
commit 78f6611f (2022-03-07): Fix #50: test wip
commit f6d44716 (2022-03-07): Fix #50: less repitition wip
commit c6dca51a (2022-03-08): Fix #50: less repitition
closed 2022-03-11 by bruhwiler