radiasoft/rslaser_old #55: consolidate RadiaSoft constants into the pykern repository

bruhwiler, 2022-03-14:
1. augment pykern/pkconst.py with constant declarations from the following two sources:
https://github.com/radiasoft/rsbeams/blob/master/rsbeams/rsphysics/rsconst.py
https://github.com/radiasoft/rslaser/blob/master/rslaser/utils/constants.py

Also, add a new constant value:  TWO_RT_TWO_LN_TWO = 2sqrt(2log(2))

2. update the rsbeams/ and rslaser/ repositories to use pkconst.py and remove all dependencies on the two files above

3. delete rsconst.py from rsbeams/ and delete constants.py from rslaser/

bruhwiler, 2022-03-21:
We have decided not to create a SciPy dependency within Pykern.

@robnagler suggested we move RS constants into a new base Python module with suggested name 'rsmath'

I agree with creating a new base Python module but suggest the name should be 'rsmathphys'

gurhar1133, 2022-03-21:
This would live in /radiasoft/rsmath or /radiasoft/rslaser/rsmath ?

bruhwiler, 2022-03-21:
/radiasoft/rsmath   although I'm suggesting  /radiasoft/rsmathphys

The problem we're working to resolve is that rsbeams/ and rslaser/  have both implemented independent solutions.
So the solution must be external to rslaser/

commit 23150814 (2022-03-14): Fix #55: removed utils/constants.py and now ref pykern
commit 051be903 (2022-03-15): Fix #55: consts from pkconst
commit bcb44c99 (2022-03-23): Fix #55: use rsmath.const instead of rslaser.utils.constants
commit d3eb512e (2022-03-23): Merge pull request #71 from radiasoft/55-constants-from-rsmath

Fix #55: use rsmath.const instead of rslaser.utils.constants
closed 2022-03-23 by bruhwiler