radiasoft/rslaser_old #62: create tests for CrystalSlice and LaserPulseEnvelope classes

bruhwiler, 2022-03-15:
Test by instantiation and one or two unit tests.

Also, there should be at least one test of the propagation() method for all element types: drift, lens, crystal_slice and for each propagation type.

gurhar1133, 2022-03-15:
I've added a few tests for the envelope already:

```python

def test_envelope():
    e = pulse.LaserPulseEnvelope()

def test_envelope2():
    trigger_exception_test(pulse.LaserPulseEnvelope, 1)

def test_envelope3():
    trigger_exception_test(
        pulse.LaserPulseEnvelope,
        PKDict(
            test='test'
        )
    )

```

which LaserPulseEnvelop methods should I add tests for (if any)?

bruhwiler, 2022-03-15:
evaluate_envelope_ex()
You can find examples of usage in utils/plot_fields.py

gurhar1133, 2022-03-15:
Is the return type for .evaluate_envelope_ex() supposed to be  ``` <class 'numpy.ndarray'> ```?

bruhwiler, 2022-03-15:
It can be either, depending on the arguments.

commit d9a65ed1 (2022-03-16): Fix #62: test reorganize
commit 500fc2f2 (2022-03-16): Fix #62: exceptions for undefined propagation + tests
commit 9b654271 (2022-03-16): Fix #62 element and envelope tests WIP
commit 9f31de62 (2022-03-16): Fix #62: whitespace
closed 2022-03-16 by bruhwiler