radiasoft/rslaser_old #68: update docstrings for LaserPulse, Crystal and other classes

bruhwiler, 2022-03-21:
The docstrings should explain the meaning and units of all PKDict parameters.

class LaserPulse(ValidatorBase):
    """
    The LaserPulse contains a LaserPulseEnvelope object to represent the initial envelope,
    as well as an array of LaserPulseSlice instances, which track details of the evolution in time.

    Args:
        params (PKDict):
            required fields:
                phE (float): photon energy [eV]
                nslice (int): number of slices
                chirp (float): total photon energy shift from 1st to last slice [eV]
                w0 [m]
                a0

gurhar1133, 2022-03-21:
See [style guidelines](https://google.github.io/styleguide/pyguide.html) for radiasoft documentation (we follow google python style guide). If you command + f on 'docstring' you will see some examples.

bnash, 2022-03-21:
I note that 
z_waist
z_center
d_to_w
all represent the purported distance from the pulse center to the waist.
We should check how [these](https://github.com/radiasoft/rslaser/blob/master/rslaser/pulse/pulse.py#L74) should be set consistently and potentially remove some of these as input parameters.

bnash, 2022-03-22:
I created the branch 
68-update-docstrings-for-laserpulse-crystal-and-other-classes
I then executed the following commands in the `rslaser` repo in Jupyter
`git fetch origin`
`git checkout 68-update-docstrings-for-laserpulse-crystal-and-other-classes`

I then made some changes to two files, git added them, committed them and pushed them.
How do I create a pull request from this? (when ready)

robnagler, 2022-03-22:
Visit the main page of the repo: https://github.com/radiasoft/rslaser  It will offer to make a pull request.

bnash, 2022-03-22:
Looks like I somehow failed to switch to that branch and pushed to master.

commit a7043e70 (2022-03-22): Fix #68: improve docustrings for several classes and functions
commit 272a50d1 (2022-03-22): Fix #68: edit docstrings
closed 2022-03-22 by bruhwiler