radiasoft/rslaser_old #69: consider and clarify use of numSig, chirp and related parameters in LaserPulse and LaserPulseSlice

bruhwiler, 2022-03-21:
Consider renaming numSig to num_sig_cutoff or sigma_cutoff

bnash, 2022-03-23:
numsig should be moved from slice defaults to pulse defaults.
Slice will then see the value of numsig.
@gurhar1133

bnash, 2022-03-23:
In laserpulse, set self.sig_s = params.tau_fwhm * const.c / 2.355
This will allow use of sig_s for a laser pulse in the wave front sensor class.

gurhar1133, 2022-03-23:
> numsig should be moved from slice defaults to pulse defaults. Slice will then see the value of numsig. @gurhar1133

+ change numsig to sigma_cutoff

bnash, 2022-03-23:
Considering the issue with the energy chirp, it looks to be adequate as is. The change in energy per slice _dE is set by the total chirp/numslice.
We can revisit this later if another way of setting chirp is preferable.

commit 2aafcb15 (2022-03-24): Merge pull request #72 from radiasoft/69-num-sig-fix

Fix #69: numsig now sig_cutoff and sig_s set as LaserPulse attr
commit ffba9bfd (2022-03-24): Fix #69: numsig now sig_cutoff and sig_s set as LaserPulse attr
closed 2022-03-24 by bnash