radiasoft/rslaser_old #74: Tests for WavefrontSensor class

gurhar1133, 2022-04-05:
* Rename as WaveFrontSensor 

## Tests:
* Basic instantiation
* Some basic input validation (distance_from_pulse_center should be float)
* Propagate() should validate that input is a laser_pulse
* IN: 1 slice and distance_from_pulse_center = 0, OUT: laserPulseSlice.wfr should remain the same
* instantiate wavefrontsensor, propagate and check output
* same as above +  confirm output type is wavefront

lmk if more @bnash

bnash, 2022-04-07:
Thanks, @gurhar1133. It would be good to know where in the electric field array that parameter is.
In any case, it's a little disturbing that the wavefront is changing so much when propagating zero length.
I'll try to understand what's going on.

gurhar1133, 2022-04-07:
I'll add a reference to where in the electric field that diff is > epsilon in the error message

robnagler, 2022-04-08:
use pkfail, not assert. if the function is private, begin with underscore.

commit 947d9541 (2022-04-05): Fix #74 WIP
commit 28aa394e (2022-04-06): Fix #74: docs + refact + tests WIP
commit dde17532 (2022-04-06): Fix #74: small cleanup
commit 9a765d66 (2022-04-06): Fix #74: test compares data vs work
closed 2022-04-08 by bruhwiler