radiasoft/rslaser_old #86: Re NLThermalSaturation.ipynb: Resolve factor of 2–3.

dtabell, 2022-09-21:
In earlier work—see our [IPAC’22 paper](https://dx.doi.org/10.18429/JACoW-IPAC2022-THPOTK062)—we identified a factor of order two or three difference between our FEniCS simulation results and the LBNL experiments. And on the basis of that result, we concluded that such simulations require a better knowledge of the true thermal properties of the crystal laser amplifiers of interest.

I believe the above conclusion remains valid, but in the process of implementing (and _debugging_) the nonlinear heat equation, I think I may have resolved that factor. Check this!

_NB:_ Olivier Chalus pointed me to [Thomas Planchon’s PhD dissertation](https://www.hal.inserm.fr/X-LOA/tel-00005388), and to, in particular, data c. p73 _ff_.

bruhwiler, 2022-10-25:
In the recent presentation to the collaboration meeting @nickgoldring showed that there is one data source with two values for thermal conductivity and specific heat which would resolve this mysterious factor.

There are other information sources, which lead to different factors.

bruhwiler, 2023-01-03:
This has been resolved.

closed 2023-01-03 by bruhwiler