radiasoft/rslaser_old #87: Re NLThermalSaturation.ipynb: Compare results produced by the linear and nonlinear heat equations.

dtabell, 2022-09-21:
In particular:
- [ ] How do they compare near room temperature?
- [ ] At what power level does one see significant differences?
- [ ] How do they compare at cryogenic temperatures?
- [ ] At what power level does one see significant differences at those lower temperatures?

bruhwiler, 2022-10-04:
@dtabell said this could be done by Wednesday, October 5

dtabell, 2022-10-11:
At room temperture, @nickgoldring and I see only very small differences.
We see those differences only by looking at specific numerical values.
Graphs of the data appear indistinguishable to the eye.

dtabell, 2022-10-11:
We increase the power level by a factor of five—from 20 W to 100 W—and the temperature rise at the entrance face differs by slightly more than 1°C.

bruhwiler, 2022-10-11:
This is good news. Does this mean that the factor of 3 confusion with the previous linear simulations has been resolved?

dtabell, 2022-10-11:
I think yes. But not committing myself just yet.
Here we simply modified the thermal diffusivity function to return a constant.

bruhwiler, 2023-05-01:
@dtabell -- please recreate this issue in rslaser-examples/ and designate priority 2
After doing so, please close this issue.