radiasoft/rslaser_old #134: Integrate thermal modeling into crystal class

morhenderson, 2023-01-19:
Update the crystal class to include all thermal modeling, up to the production of ABCD matrices. Tasks:
- [ ] Add physical crystal parameters as arguments to initialization routine
    - [ ] Include crystal geometry & add a 3D mesh class member
- [ ] Add thermal simulation method to crystal class
    - [ ] Allow passing of simulation parameters as a dictionary 
    - [ ] Add spatial evaluations to simulation routine (with points as a simulation parameter) 
- [ ] Create dictionaries of physical crystal parameters for multiple types
    - [ ] Add import-able dictionaries to "crystal" submodule
- [ ] Allow/demonstrate time resolution for all physical properties (incl. temp, index of refraction, ABCD matrices)
- [ ] Account for separation of crystals into slices
- [ ] Create a notebook to demonstrate new functionality

bruhwiler, 2023-05-01:
@morhenderson -- what is the status of this effort?

There have been a number of changes to the crystal class, so it might be necessary to reconsder some of the tasks.

morhenderson, 2023-05-01:
Yes, given the major updates that were occurring I just shelved what I had done for this effort. If the main repo is in a more stable configuration now I can start updating these plans to match what was done there.

bruhwiler, 2023-05-01:
The rslaser repo is getting updated regularly.
When you get started, it would be good to create more atomic issues, so the corresponding branch/PR can be short lived.

bruhwiler, 2023-05-01:
Issues should be created in the 'new' rslaser/ repo.