radiasoft/rslaser_old #136: Adding gain calculation to crystal slice propagate method

k-wolfinger, 2023-01-23:
Need to restructure CrystalSlice.propagate() method to logically accommodate gain calculation.

bruhwiler, 2023-01-23:
We have multiple prop types in the crystal slice.
More than one of these (perhaps not all) will require gain calculations.
I think we should allow users to toggle gain calculations on/off

The propagate method needs to be restructured to accommodate this complexity, without a lot of copy/paste and annoying if blocks.

bnash, 2023-01-23:
Yes, I'm not sure the best way to structure that logic. I would suggest creating a new propagator that just does the gain and then also one that does both the linear propagation (via SRW or LCT) and the gain.
[Note](https://github.com/radiasoft/rslaser/blob/master/rslaser/optics/element.py#L165) propagator 'n0n2_lct' for an example of a propagator that doesn't use SRW. Something like this needs to be done for the gain propagator also.

gurhar1133, 2023-01-24:
Let me know how I can assist with all of this

commit 27cbd236 (2023-01-27): Fix #136: Whitespace
commit b5920dee (2023-01-27): Fix #136: merge fix
commit 4564f634 (2023-01-27): Merge pull request #140 from radiasoft/136-refactor-propagation

Fix #136: Refactor propagation
commit e15f09c6 (2023-01-27): Fix #136: WIP
closed 2023-01-27 by bruhwiler