radiasoft/rslaser_old #147: Compare gain simulation with experiments

k-wolfinger, 2023-02-08:
Compare with the amplified Berkeley experiments (look at LaserPulseInit01, SinglePass01)
    1. Read in peak timing files for comparison with simulation (ccd intensity, wfs phase)
    2. Use the pump_off case as the initial wavefront, propagate through gain calculation, compare simulation output with the peak timed data

Will need to combine wavefronts to properly compare with wavefront sensor data.

k-wolfinger, 2023-02-10:
Also:

Look at other dependencies for gain, like mesh resolution (nx, ny)
Initialization types for the crystal at some point: flat, exponential decay (1 vs 2 sides), outside data

bruhwiler, 2023-02-15:
The WFS diagnostic creates a file with the 2D transverse intensity profile, integrated longitudinally over the laser pulse.

This data is provided in 'arbitrary' units.

Hence, we need to use the energy of the 800 nm seed pulse to correctly convert the WFS intensity to physical units.

closed 2023-02-22 by bruhwiler