radiasoft/rslaser_old #152: generalize file-based constructor of LaserPulse for multiple slices

bruhwiler, 2023-02-15:
The initial implementation is documented in #82 

The present implementation will not work for multiple laser pulse slices.

k-wolfinger, 2023-02-17:
Add 'one' laser wavefront from data files to multiple laser slices by dividing fields by number of slices

- will eventually need gaussian longitudinal profile, maybe not this issue

Also, need to make changes to crystal.py (just added them to this issue):

- need to set self.A = param.A, etc for use with _propagate_abcd_lct
- changes to mesh by lct propagators causes issue with self._interpolate_pop_change() when propagating through multiple crystal slices, check and fix

k-wolfinger, 2023-02-17:
Can't find a description of the `files` object for formatting when passing to `LaserPulse` constructor. I'll try to write it based on how `files` is used.

k-wolfinger, 2023-02-21:
When the propagators change the size of wfr.mesh, they don't affect the size of the n_photons_2d mesh. This can cause the gain calculation to fail.

I will fix this by interpolating the mesh of photon numbers to match wfr.mesh, within the gain calculation.

To make this easier, I'll combine _interpolate_pop_change and _interpolate_pop_inversion into a generic _interpolate_a_to_b(a,b) like function.

k-wolfinger, 2023-02-23:
@gurhar1133 I made changes to `_wavefront()` in pulse.py so that it can initialize a pulse from file with multiple laser slices. This is causing tests/pulse_test.py::test_from_file to fail. Can you take a look at it?

gurhar1133, 2023-02-23:
> @gurhar1133 I made changes to `_wavefront()` in pulse.py so that it can initialize a pulse from file with multiple laser slices. This is causing tests/pulse_test.py::test_from_file to fail. Can you take a look at it?

Sure 👍  what branch are the changes on?

k-wolfinger, 2023-02-23:
> > @gurhar1133 I made changes to `_wavefront()` in pulse.py so that it can initialize a pulse from file with multiple laser slices. This is causing tests/pulse_test.py::test_from_file to fail. Can you take a look at it?
> 
> Sure 👍 what branch are the changes on?

Great! They're on the one linked to this issue, 154-file-based-multiple-laser-slices

gurhar1133, 2023-02-23:
> Great! They're on the one linked to this issue, 154-file-based-multiple-laser-slices

Ok, I'll take a look

closed 2023-03-02 by bruhwiler