radiasoft/rslaser_old #156: resolve propagator issues

k-wolfinger, 2023-02-22:
Currently not getting good phase output of propagators + changes to mesh dimensions and extent.

k-wolfinger, 2023-03-03:
To do

Write a function that wraps the propagators and:
- Makes sure output is 2^n
- Improve padding: 
1. pad, smooth, propagate... 
2. use windowing
3. 'automate' how many zero cells between edge and the first non-zero value?
- Truncates the wfr mesh if it is larger than a factor times the crystal radius
- Check dx and dy against a specified minimum (also not greater than ~0.1x crystal radius?)

k-wolfinger, 2023-03-08:
Before closing:
* put in phase cut

k-wolfinger, 2023-03-09:
@bruhwiler Two notes before creating the pull request:
1. Because of the windowing applied to the file-initialized fields, we now get fewer photons in file-init than in function-init (where previously they were the same)
2. We assume electric_field_y is zero in the file-init and other places. Should we change this?

k-wolfinger, 2023-03-09:
New changes:
- Still pad 64x64, move windowing to the intensity file-data, try padding with a gaussian fit, 
- Cut the mesh during propagation but don't touch phase
- Cut phase when plotting if needed

bruhwiler, 2023-03-09:
Let's continue assuming E_y is zero for now.

closed 2023-03-10 by bruhwiler