radiasoft/rslaser_old #159: _propagate_n0n2_srw() not calling PropagElecField() when n2 == 0

moellep, 2023-03-03:
https://github.com/radiasoft/rslaser/blob/7edbf2888c24272ac1eba3d06a10eeeba9b521bc/rslaser/optics/crystal.py#L431

In ```crystal.py _propagate_n0n2_srw()``` if n2 is 0, it creates an optBL container structure, but then does nothing with it. It looks like the call to ```srwlib.srwl.PropagElecField(thisSlice.wfr, optBL)``` should be out-dented a level so the optBL is propagated for all values of n2.

bruhwiler, 2023-03-03:
@k-wolfinger -- please have a look.
It would be worth checking the other prop_type methods as well.

Also, we should check for n2<0 and throw an exception in that case.

k-wolfinger, 2023-03-03:
* `_propagate_n0n2_lct` is fine
* `_propagate_abcd_lct` is fine
* `_propagate_n0n2_srw`:
    * Had to out-dent `srwlib.srwl.PropagElecField(thisSlice.wfr, optBL)` a level
    
Testing changes and adding a check for n2<0 now

k-wolfinger, 2023-03-03:
Added check for n2<0 and tested changes. Tests all passing.

closed 2023-03-03 by bruhwiler