radiasoft/rslaser_old #179: Repo is bloated: 147MB

robnagler, 2023-03-21:
It's important for Python repos to be small (<10MB) so that image builds can be fast. Cloning rslaser now takes many minutes, because the compressed git repo is 147MB. The majority of this is taken up by the examples directory. 

The fix would be to rename rslaser to rslaser-old. Copy in the necessary files, and recreate the repo. I suspect it would be too complicated to edit the version repo.

Research repos should be named something else, e.g. rslaser-research or rslaster-notebooks. The notebooks that are saved to git should have their plots cleared. Several notebooks in examples are over 10MB. 

It may not make sense to do anything right now. Just noting here, because if ever rs repo were to take 3 minutes to download, our image builds would take 30 minutes longer (at least) than they need to.

bruhwiler, 2023-03-22:
I'll rename it to rslaser-old

bruhwiler, 2023-03-24:
rslaser-examples has been created, so people can begin to clean up / prune  and then move notebooks to that repository

rslaser-new has been created, so the Python modules, tests, etc. can be moved over there for testing.

closed 2023-03-30 by bruhwiler